kusti8/hue-plus #2: hue.py freezes

epsol, 2016-10-21:
I am running into issues where if I issue any command via the hue/hue.py that it will just freeze until the system has been up about 5 minutes or so.  Is there anything that can be done to reduce this delay?

kusti8, 2016-10-22:
Huh. I have no idea what could cause this. It may be that the system needs some time until it can establish a connection with the serial device. I'm not sure and it probably depends on the distro, not the program.

epsol, 2016-10-28:
Actually I have more info on this, it doesn't seem like a delay, I just have to run the hue.py 8 times before it starts responding... seems to do it both on ubuntu and cent.... just curious if you might have some idea as to why?

kusti8, 2016-10-29:
I have no idea what could cause this. Are the lights white or the previous color? Any strange output?

epsol, 2016-10-29:
It doesn't seem to matter, running previous color vs completely cutting the power and having it start up white.

kusti8, 2016-11-10:
I've pushed a new release which should fix the problem and also saves your previous settings so you can change channels easily. Please comment if it persists. Closing.

closed 2016-11-10 by kusti8

epsol, 2016-11-11:
Just wanted to reply that I really have no way of testing this.  A week ago under windows it performed a firmware update and after that every time I run the hue.py(both old and the one you released today) it just locks the hue+ to the point where I have to completely pull the power to even have windows pick it up again.

kusti8, 2016-11-11:
I have no idea. Does CAM work correctly?

epsol, 2016-11-11:
yes, cam under windows seems to work.  but if I reboot into linux, run the hue.py, have the lights freeze, reboot back into windows then it doesnt detect.  I have to pull the power cord from the machine in order to get it to show up under cam again.  Also on a side note,, with this new firmware, on a complete power loss, the lights do not turn on white, they just stay off until the cam software is loaded.  

I know its nothing to do with your script.... its definitely something to do with this new firmware pushed a couple days ago.

epsol, 2016-11-11:
Also not sure if it helps or not, but it appears they added a new mode called 'Wings'

kusti8, 2016-11-11:
I'll look into it.