kusti8/hue-plus #4: Hue script appears to hang.

garwil, 2016-12-27:
When I run any of the commands listed in the readme, nothing happens and the command has to be terminated with Ctrl-C. My Hue+ device is attached to an internal USB2 port on the motherboard and is listed as being on /dev/ttyACM0. I have tried running the commands using sudo and also from a root shell, but get the same result. The device works under Windows 10 but I no longer have this installed.

From a glance at the source, hue.py doesn't appear to generate log files, but I get the following output when I terminate the script.
```
sudo hue fixed ff0000

^CTraceback (most recent call last):
  File "/usr/bin/hue", line 295, in <module>
    fixed(ser, args.gui, args.channel, args.color)
  File "/usr/bin/hue", line 79, in fixed
    ser.read()
  File "/usr/lib/python3.5/site-packages/serial/serialposix.py", line 472, in read
    ready, _, _ = select.select([self.fd, self.pipe_abort_read_r], [], [], timeout.time_left())
KeyboardInterrupt
```

I am running Antergos Linux (based on Arch) using the default kernel. Please let me know if there is any other information you need.

epsol, 2016-12-29:
if you have updated the firmware in your hue then that is why it's not working.  NZXT changed the protocols enough where this script no longer works.

garwil, 2016-12-31:
Thanks for your reply. Have they changed the protocols enough that it will never be able to work? If not, is there any way I can assist in getting it working again?

epsol, 2016-12-31:
I think the developer plans to work on it again, just short on time.

garwil, 2016-12-31:
Awesome, I'll keep an eye out then!

chrisromic2, 2017-01-07:
Does this affect newer versions of the Hue+, or does it only happen when the user updates the firmware themselves?

kusti8, 2017-01-13:
Meaning to put this, but I've been working on the new protocol on the dev branch. I'm still trying to debug an issue with single channels and I don't support backwards marquee and cover_marquee but everything else will work just fine. The code is also cleaned up a bit and more portable.

kusti8, 2017-01-13:
@chrisromic2 I think only when updated but I'm not sure of they're shipping units that use it.

epsol, 2017-01-14:
I gave the dev branch a try, but I seem to keep getting 'IndexError:  string index out of range.'

kusti8, 2017-01-14:
Where? What's the full output? The latest version I tested worked for me
but I may have accidentally done some changes

epsol, 2017-01-14:
this is from hue power off:

Traceback (most recent call last):
  File "/usr/bin/hue", line 295, in <module>
    power(ser, args.channel, args.state)
  File "/usr/bin/hue", line 269, in power
    fixed(ser, 0, channel, "000000")
  File "/usr/bin/hue", line 139, in fixed
    command = create_command(ser, channel, [color], "fixed", 0, 0, 0, 2)
  File "/usr/bin/hue", line 86, in create_command
    strips = [0, strips_info(ser, 1)-1, strips_info(ser, 2)-1]
  File "/usr/bin/hue", line 117, in strips_info
    return int(ser.read(ser.in_waiting).hex()[-1])
IndexError: string index out of range

qdel, 2017-01-15:
Hi,

Same, it seems the hue never answer to the command. (i tried upping the sleep, in case of...)

jcorrado, 2017-01-16:
@chrisromic2 I purchased my HUE+ from Amazon on 2017-01-07 and it works correctly with the current HEAD of master (3bccdea).

@kusti8 thanks for building this!  Great to have these LEDs working on my Debian system, at home.

kusti8, 2017-01-16:
Yeah I can recreate the issue. They may have changed the protocol again, I'll look into it.

kusti8, 2017-01-16:
That's weird. I opened CAM up, then closed it and tried this again and it suddenly worked.

qdel, 2017-01-17:
Ok, i will give this a try. But i need to use a old laptop because this pc doesn't have windows and the CAM software do not work on virtual machine...

Which cam version? I just start, play with it and close it propertly?

kusti8, 2017-01-17:
I used the newest version

qdel, 2017-01-18:
I tried with the hue plug on a windows laptop. It is correctly working.

But when i plug it back to linux and run the script... same issue...

qdel, 2017-02-03:
I managed to make it work.... Using an uber pretty shell command:

` R=1; while [ $R -eq 1 ] ; do sudo ./hue.py power off; R=$? ; done`

Also, i needed to make a little correction on the strips_info method. It could return '0'. When used inside create_command function, it will be -1.

The bytearray cannot use a '-1'.

This seems to happen because i use only 1 strip and not 2.

kusti8, 2017-02-03:
I'll look into it. Can you outline your correction to the script? I'm terribly busy and I hope I can get some time to work on it, but I may not. A PR would be nice if you could.

qdel, 2017-02-04:
Done.

Also, today when powering up the pc, the script was hanging.
Using my magic uber pretty shell command restore functionality after a ~10 seconds.

I think my pc send dirty bytes to com port. Flooding it makes the hue find a new "start".

KevOrr, 2017-02-05:
I'm getting the same `IndexError`. All I can say is that in `strips_info()`, `ser.read()` returns `b''`, and in wireshark, the response packet has an empty data section

Cyriis, 2017-02-07:
I tested the fix from @qdel but I think it raised the same error as with @KevOrr:

> Traceback (most recent call last):
>   File "./hue.py", line 278, in <module>
>     fixed(ser, args.gui, args.channel, args.color)
>   File "./hue.py", line 142, in fixed
>     command = create_command(ser, channel, [color], "fixed", 0, 0, 0, 2)
>   File "./hue.py", line 86, in create_command
>     strips = [0, strips_info(ser, 1)-1, strips_info(ser, 2)-1]
>   File "./hue.py", line 117, in strips_info
>     r = int(ser.read(ser.in_waiting).hex()[-1])
> IndexError: string index out of range

qdel, 2017-02-08:
This is normal, my fix isn't in any way intended to fix this issue...

There is 2 bugs in the script:

- problem with the bytearray creation failing because a -1 is reported by the hue because i use only 1 strip of leds. My PR is for this bug, as described in the commit message "Correcting script when using only one line".
- The hue seems not to answer when requested. To fix this problem, i use my shell command line.

kusti8, 2017-02-09:
I think I've fixed all the issues. Could people try them and see how it works before I merge? It works for me.

nesslie, 2017-02-10:
I tried the new firmware, works just fine now on Kubuntu 16.10. Thanks!

Cyriis, 2017-02-10:
Tested on my side, newest firmware. Works at the first try (Arch Linux).

kusti8, 2017-02-10:
Ok, I'll merge it soon

closed 2017-02-10 by kusti8

garwil, 2017-02-13:
Thank you so much! I'm so grateful!

PR #6 (2017-02-10): Add new firmware support
This is pretty much the final version unless any extra bugs are found. Let me know here of any issues. Fixes #4