kusti8/hue-plus #8: WebColors

keyxmakerx, 2017-03-30:
I have already install webcolors. 

keyxmakerx@Emerald:~/Downloads/hue-plus-master$ sudo -H pip install webcolors
Requirement already satisfied: webcolors in /usr/lib/python2.7/dist-packages

keyxmakerx@Emerald:~/Downloads/hue-plus-master$ sudo hue-ui
Traceback (most recent call last):
  File "/usr/bin/hue-ui", line 14, in <module>
    import webcolors
ImportError: No module named 'webcolors'
keyxmakerx@Emerald:~/Downloads/hue-plus-master$

kusti8, 2017-03-30:
You must install Python 3 version. sudo - H pip3 install webcolors

keyxmakerx, 2017-03-30:
This did resolve the issue, my apologies for not thinking of that sooner. 

Probably no way of trying to figure out which usb port it's in is there?

kusti8, 2017-03-30:
It's almost always in the default port /dev/ttyACM0. Try that and see if it works. If you really want to know then just check:
ls /dev/ttyACM*

closed 2017-03-30 by kusti8

keyxmakerx, 2017-03-31:
Confirmed it is working. Thank you again for your help!