kusti8/hue-plus #9: Collaboration to add more NZXT devices and create a cross platform alternative to CAM.

dkanada, 2017-04-14:
Perhaps you could talk to the deveopler for [this](https://github.com/akej74/grid-control) project and try and add more devices to this program. I have also found an [application](https://github.com/jaksi/leviathan) that can customize the Kraken devices, it looks like the [protocol](https://github.com/jaksi/leviathan/blob/master/PROTOCOL.md) used by the X61 is also listed in that repository. I have not worked with python before but I have an X61 myself so I would love to contribute to an alternative for CAM that works on Linux.

kusti8, 2017-04-16:
It is being worked on right now.

lolrepeatlol, 2017-07-02:
+1. I'd like my H7 Quad Lumi to work.

Alienmoon, 2017-07-10:
@kusti8
Found your comments over on NZXT, downloaded and will keep in touch, I totally appreciate your work.

There is another guy on here who made an alternative app for the NZXT Grid+, and he did also start one for the HUE+ and Kraken's, however, he stopped development because of issues with the not knowing the USB Protocols or something like that.

Thanks

akej74, 2017-07-10:
@Alienmoon 
Hi, I'm the developer of "Grid Control" (linked in ther first post above) and I did some testing with USB protocols, but as I have no Kraken HW to test with it was a bit difficult. I also run into a wall of problems with the USB communication and had to give up further development for supporting Kraken devices for now.

kusti8, 2017-07-10:
Without having a lot of money and resources, it's impossible to develop a full alternative for CAM. NZXT can do it cause they make the products, it their products cost a lot so I can't get them to test with.

JezzupWuzzup, 2017-07-15:
We need an alternative for our NZXT devices. Those with NZXT coolers should no longer use the first party software of course due to the software uploading information about its users.

Alienmoon, 2017-07-15:
@akej74 
Hey, Pal, Yes, I recognise your name and I was sure I was following you and watching your projects, just in case something changes, some reason the site claimed I'd unfollowed you for some reason.
moving on, your work is also appreciated, Thanks to both of you.

ChadAlanJackson, 2017-09-13:
Now that Asus Aura has published an SDK, it would be amazing to add their devices into this mix for a more unified solution across a couple of brands.

In all seriousness, what you've done here is fantastic. Keep up the good work.

kusti8, 2017-09-13:
@ChadAlanJackson Asus Aura is written in c++ and requires Windows. I'm not going to support it.

MrBlue1227, 2019-02-20:
Hi, Habt ihr schon eine Lösung wie die Software den LED Hub erkennt

closed 2020-01-12 by dkanada