kusti8/hue-plus #12: Can't apply affects.

cbillingham465, 2017-04-16:
Serial port is invalid. "Try /dev/ttyACm0 for Linux or COM3 or COM4 for Windows". It is a new system as of 4/13/17 and I haven't reactivated Windows so I don't if either of those would cause an issue. That is what pops up when I try to apply any affect.

cbillingham465, 2017-04-16:
Guy was right in the comments on reddit. I just had to turn off CAM and then restart. Although when I load CAM back up the lights get disabled again and I do actually like the visual layout of the CAM temp gauges and stuff.

kusti8, 2017-04-16:
Cam does disable the lights so you have to stop it from running. I'm not going to focus on looks since I don't have a lot of time anyone. I'm focusing on functionality and adding new features.

closed 2017-04-16 by kusti8