kusti8/hue-plus #16: AER RGB

lolrepeatlol, 2017-08-11:
Does this work with AER fans? 
ty

PS: Sorry for not donating anymore, couldn't afford it. :/

kusti8, 2017-08-11:
It should, yes.

Alienmoon, 2017-08-11:
I have most of the NZXT devices, HUE+, AER RGB Fans and the Kraken x62 along with the NZXT USB internal HUB v2.

I'd like to ask if this is intended to work with all NZXT RGB devices "Kraken x62" or is it just the AER RGB Fans and the HUE+?

I'm also open to help test this software if you do intend for it to work with all NXT devices and need feedback and help to make this happen?

Thank you for the work your doing Pal, I'll be relieved to do away with this CAM software, all NZXT has done is make it onto a piece of bloatware, all I wanted was to control the lighting effects.

kusti8, 2017-08-13:
The Kraken is not even remotely close to the same protocol as the Hue+ and I don't have one to test with. Until I can get one, there's no chance of me getting it working.

closed 2017-08-13 by kusti8

Alienmoon, 2017-08-13:
@kusti8
That's not what I said,  I said ALL the RGB Kraken coolers will be the same protocol because they all have the same RGB cooling block and basically all do the same thing with RGB ring and text.

I understand perfectly that the HUE+ and AER Fans are totally different protocols, I'm not too sure why you thought I was claiming there the same as the HUE+, I'm guessing you miss read what I said,... No worries.

kusti8, 2017-08-13:
They should be the same, but they aren't. I don't know what NZXT did, but
the v1 and v2 are different protocols.