kusti8/hue-plus #17: dev-python/sip installed, but claims couldn't find version match

jcfrosty, 2017-08-18:
jcfrosty@CtrlZ ~ $ sudo pip3 install --user hue_plus      
Collecting hue_plus
  Using cached hue_plus-1.4.1.tar.gz
Collecting pyserial (from hue_plus)
  Using cached pyserial-3.4-py2.py3-none-any.whl
Collecting pyqt5 (from hue_plus)
  Using cached PyQt5-5.8-5.8.0-cp34.cp35.cp36.cp37-abi3-manylinux1_x86_64.whl
Collecting pyaudio (from hue_plus)
  Using cached PyAudio-0.2.11.tar.gz
Requirement already satisfied: appdirs in /usr/lib64/python3.4/site-packages (from hue_plus)
Collecting sip>=4.19.1 (from pyqt5->hue_plus)
  Could not find a version that satisfies the requirement sip>=4.19.1 (from pyqt5->hue_plus) (from versions: )
No matching distribution found for sip>=4.19.1 (from pyqt5->hue_plus)

jcfrosty@CtrlZ ~ $ eix dev-python/sip
[U] dev-python/sip
     Available versions:  4.19.1(0/12) (~)4.19.2(0/12) (~)4.19.3(0/12) **4.9999(0/12) {debug doc PYTHON_TARGETS="python2_7 python3_4 python3_5 python3_6"}
     Installed versions:  4.19.1(06:08:03 AM 04/30/2017)(-debug -doc PYTHON_TARGETS="python2_7 python3_4 -python3_5 -python3_6")
     Homepage:            https://www.riverbankcomputing.com/software/sip/intro
     Description:         Python extension module generator for C and C++ libraries

kusti8, 2017-08-23:
This isn't an issue with hue-plus, but with your setup and pip. For some reason it can't find the sip package. I would look at what python versions you have installed and where they are.

closed 2017-08-23 by kusti8

jcfrosty, 2017-08-26:
jcfrosty@Silent-Knight /opt $ eselect python list
Available Python interpreters, in order of preference:
  [1]   python3.4
  [2]   python2.7
  [3]   python3.4
jcfrosty@Silent-Knight /opt $ eselect python show
python3.4
jcfrosty@Silent-Knight /opt $ which python
/usr/bin/python
jcfrosty@Silent-Knight /opt $ python --version
Python 3.4.5

jcfrosty, 2017-08-26:
jcfrosty@Silent-Knight /opt $ sip -V
4.19.1

jcfrosty, 2017-08-26:
I've even installed the requirements without pip. so how do I compile this and install it manually?