kusti8/hue-plus #18: Audio mode low sample rate, latency

thundR, 2017-08-23:
Was hoping that this project would have better audio support than CAM, but the audio mode retains the laggy mess of CAM control. Any hope of a fix?

kusti8, 2017-08-23:
It works perfectly fine for me and also on CAM. I would try changing the rate to something lower. Can you take a video or something? I don't exactly understand what the problem is here.

thundR, 2017-08-24:
I'll try to describe the problem, one that many, many people have had in the past with CAM as I've seen on the NZXT subreddit and other posts. 

The audio mode has a latency effect of 1-2 seconds from the actual audio source. Furthermore, the refresh rate seems incredibly sluggish at only a couple times a second. The result has basically nothing in common with the audio it's working with. It seems weird to me, cause FFTs with LEDs are probably one of the most commonly completed projects for these types of things. 

I guess I should also establish what "perfectly fine" is, since everywhere I've read, it seems pretty universal that the audio mode is non-functional, at least in CAM. I remember one post saying there's only one video on youtube with the software working properly. 

If it looks great on your end, it could be something like the fact that my audio interface uses proprietary drivers, or some other issue. I'll give the code a look.

Alienmoon, 2017-08-24:
@thundR  agreed, this is the only reason I don't use the audio, It just seems out of time.

kusti8, 2017-08-24:
Ok I'll look into it. Did you try lowering the smoothness?