kusti8/hue-plus #20: Save last setting

aur0n, 2017-10-06:
Hi, is there anyway to make hue_plus start the last mode? Everytime I open it, I have to manually select the desired mode.

I'd like to make run in on startup and having the leds automatically light up.

Thank you.
PS: I'm on Windows 10.

kusti8, 2017-10-21:
It should automatically do that. What settings are you using?

Neopolitan2D, 2017-10-27:
Hello Kusti8, I am having an issue where the LEDs aren’t active upon startup; however, once hue-plus is open they start up and go to the last saved effect. If the computer shuts down, or goes to sleep, then the LEDs will shut off until the program has been reopened. 

It was working flawlessly prior to the most recent update; but I doubt that is what’s causing the issue- I’ve tried deleting and reinstalling, to no avail. I’ve also tried deleting and going back to a previous version with no results- if there is anything I could provide to diagnose the issue, I’ll glady do my part!

kusti8, 2017-10-27:
Hue plus isn't set to automatically start up on boot. I designate this to
the user, since I personally have software that automatically starts on
boot. You can easily set it to run automatically. An example:
http://www.thewindowsclub.com/make-programs-run-on-startup-windows

I'll see if I can add it to the readme to point this out.

Neopolitan2D, 2017-10-27:
I appreciate the quick reply! And totally my bad on making a quick mistake, thank you very much

closed 2017-10-28 by kusti8