kusti8/hue-plus #21: hue_plus crashes

CartoonKnite, 2017-10-08:


CartoonKnite, 2017-10-08:
[image]
[image]

KnightSirius, 2017-10-19:
Exact same issue here!

akej74, 2017-10-20:
Not sure it this is relevant, but I hade to rewrite the "excepthook" as follows to manage unhandled exceptions in QT in a nice way:

In my "helper.py":
```
def excepthook(excType, excValue, tracebackobj):
    """Rewritten "excepthook" function, to display a message box with details about the exception.
    @param excType exception type
    @param excValue exception value
    @param tracebackobj traceback object
    """
    separator = '-' * 40
    notice = "An unhandled exception has occurred\n"

    tbinfofile = io.StringIO()
    traceback.print_tb(tracebackobj, None, tbinfofile)
    tbinfofile.seek(0)
    tbinfo = tbinfofile.read()
    errmsg = '%s: \n%s' % (str(excType), str(excValue))
    sections = [separator, errmsg, separator, tbinfo]
    msg = '\n'.join(sections)

    # Create a QMessagebox
    error_box = QtWidgets.QMessageBox()

    error_box.setText(str(notice)+str(msg))
    error_box.setWindowTitle("Grid Control - unhandled exception")
    error_box.setIcon(QtWidgets.QMessageBox.Critical)
    error_box.setStandardButtons(QtWidgets.QMessageBox.Ok)
    error_box.setTextInteractionFlags(QtCore.Qt.TextSelectableByMouse)

    # Show the window
    error_box.exec_()
    sys.exit(1)
```

And then the following in main:
```
if __name__ == "__main__":
    # Use a rewritten excepthook for displaying unhandled exceptions as a QMessageBox
    sys.excepthook = helper.excepthook
```

kusti8, 2017-10-20:
I'll add it and release a new version soon. I was planning on doing it but
got caught up in things

kusti8, 2017-10-21:
New version released. Can you run it and show what the error is?

CartoonKnite, 2017-10-28:
Hi,

Now it gives me this error (after updating)
[image]

kusti8, 2017-10-28:
How many strips do you have ? Is your device firmware up to date ?

CartoonKnite, 2017-10-31:
I have 6 strips. Just checked, and my firmware is up to date.

kusti8, 2017-11-07:
How are the strips connected ? How many on each channel?

CartoonKnite, 2017-11-07:
3 on each channel

CartoonKnite, 2018-04-02:
Still hoping this can be fixed sometime :(

CartoonKnite, 2018-07-09:
So, I had to make an upgrade. New processor, motherboard, and install of windows. I'm getting the same error still. So, I imagine that probably means my issue has something to do with the unit itself rather than something on my computer. Hope this is helpful

CartoonKnite, 2018-07-10:
Ok, update. So I have no idea why this worked, but it seems like my second port on the hue+ device can only handle two LED strips. The other one can handle three. So I just unplugged one, and everything works now.

jonkiersey, 2018-11-09:
I started running into the same problem as CartoonKnite after adding another fan. Doing some troubleshooting by isolating the new fan, it seems that an Aer RGB 2 is not compatible with the Hue+. I don't expect a fix from this project, just posting because I can't find anywhere else that lists that incompatibility and I would bet that was also what was causing CartoonKnite's issue.