kusti8/hue-plus #24: Profile/Settings File, Direct Edit with Text Editor?

jakedowns, 2017-11-26:
is it possible to edit a custom Profile directly with a hex/text editor? rather than using the GUI?

the GUI is great, but for some complex color schemes, i was hoping to be able to copy/paste my 40 colors instead of clicking through them all by hand,

or maybe even writing my own little custom scripts and/or spreadsheet functions to generate a list of colors i'd like to use for a profile.

I noticed that the profiles are saved using `settings = QSettings('kusti8', 'hue_plus')` but i couldn't find any matching files for that on disk (Windows)

Thanks for a great tool, btw, glad to be able to ditch CAM

kusti8, 2017-11-26:
It looks like it's stored in the registry under local machine/software or
current_user/software. I'm going to try and rewrite the project in go and
try and add some ideas, but that may take a while.

jakedowns, 2017-11-26:
Awesome, thanks. 

I'll go ahead and close this for now then.

Maybe I can look into call out to the CLI interface directly and using a temp file-format for generating/saving my own custom animation schemes for now.

Still would love to hear more behind how you figured out the protocol if you ever get the time

closed 2017-11-26 by jakedowns

kusti8, 2017-11-26:
I'm planning on making it a Json file that's easier to read and access. I
figured out the protocol by using a serial analyzer, and a lot of trial and
error.