kusti8/hue-plus #25: pip3 install fails on pyqt5 requirement

jakedowns, 2017-11-26:
https://github.com/kusti8/hue-plus/blob/f367bea93414fb8099109ec763d66debcbcfb202/setup.py#L31

> should the requirement be `python3-pyqt5` ?

--- 

```
> pip3 install hue_plus                                                                                                 
Downloading/unpacking hue-plus                                                                                            
Downloading hue_plus-1.4.3.tar.gz                                                                                       
Running setup.py (path:/tmp/pip_build_root/hue-plus/setup.py) egg_info for package hue-plus                                                                                                                                                   
Downloading/unpacking pyserial (from hue-plus)                                                                            
Downloading pyserial-3.4-py2.py3-none-any.whl (193kB): 193kB downloaded                                               
Downloading/unpacking pyqt5 (from hue-plus)                                                                               
Could not find any downloads that satisfy the requirement pyqt5 (from hue-plus)                                       
Cleaning up...                                                                                                          
No distributions at all found for pyqt5 (from hue-plus)                                                                 
Storing debug log for failure in /root/.pip/pip.log   
```

```
Downloading/unpacking pyqt5 (from hue-plus)
  Getting page https://pypi.python.org/simple/pyqt5/
  URLs to search for versions for pyqt5 (from hue-plus):
  * https://pypi.python.org/simple/pyqt5/
  Analyzing links from page https://pypi.python.org/simple/pyqt5/
    Skipping https://pypi.python.org/packages/01/8c/9d36bcb04f3c21bd449e438a5d52259a62dbbfe222698fa006099caf5143/PyQt5-5.8.1.1-5.8.0-cp35.cp36.cp37-none-win_amd64.whl#md5=04b2cb03efe5f544725f729eef723d28 (from https://pypi.python.org/simple/pyqt5/) because it is not compatible with this Python
    Skipping https://pypi.python.org/packages/03/03/f42a3f893c4f7b08dfc3f6187b2db5558ee3ca194180bbd8b85b8721e825/PyQt5-5.9.1-5.9.2-cp35.cp36.cp37-none-win32.whl#md5=c88436dda54af7c759fa149d76eeca87 (from https://pypi.python.org/simple/pyqt5/) because it is not compatible with this Python
    Skipping https://pypi.python.org/packages/07/7b/3223e33293be19e7666976db4ed346d83ba9b97be23aae2e4f94dcf37782/PyQt5-5.8.2-5.8.0-cp35.cp36.cp37-none-win32.whl#md5=1680375206a128d8325fb38efd784c03 (from https://pypi.python.org/simple/pyqt5/) because it is not compatible with this Python
    Skipping https://pypi.python.org/packages/0f/a8/71d918272a9ae17541942daffb15d663d7c9401346252a495a84822e81d5/PyQt5-5.8.2-5.8.0-cp35.cp36.cp37-abi3-macosx_10_6_intel.whl#md5=2b71949cfc748db399c0bd8157444a85 (from https://pypi.python.org/simple/pyqt5/) because it is not compatible with this Python
    Skipping https://pypi.python.org/packages/12/8a/83d96cef5f8155bc4eb140713bc479f56c15e542bec0c4340f65a0e8317a/PyQt5-5.7.1-5.7.1-cp34.cp35.cp36-abi3-manylinux1_x86_64.whl#md5=b4f4f015c60c6c6f0404acf3307b31c1 (from https://pypi.python.org/simple/pyqt5/) because it is not compatible with this Python
    Skipping https://pypi.python.org/packages/13/f4/dad8682ca966094fd3694746fba6b87c11ba793a4660022db7275d50b34b/PyQt5-5.8-5.8.0-cp34.cp35.cp36.cp37-abi3-manylinux1_x86_64.whl#md5=a2fdd5e3671aaef944c1a83ca1eaf60a (from https://pypi.python.org/simple/pyqt5/) because it is not compatible with this Python
    Skipping https://pypi.python.org/packages/14/60/cc8aacfc5c7fd1eb54aa91982d189b052100b2884ba7bc77b3515879a17d/PyQt5-5.9-5.9.1-cp35.cp36.cp37-none-win32.whl#md5=640473b50f6387c08fd84349793460ba (from https://pypi.python.org/simple/pyqt5/) because it is not compatible with this Python
    Skipping https://pypi.python.org/packages/21/a9/f64e3511b2e7da37447afc34d3f28d0a5b00e0cfdf5774171f9f11bfd722/PyQt5-5.9.1-5.9.2-cp35.cp36.cp37-abi3-macosx_10_6_intel.whl#md5=6bb5a992af1c817c599544f0099a0c39 (from https://pypi.python.org/simple/pyqt5/) because it is not compatible with this Python
    Skipping https://pypi.python.org/packages/26/97/b40b1052ffc676c9fb5e68cb0400309a9a22a4f582aa4294b319f3dc8b3f/PyQt5-5.8.1.1-5.8.0-cp35.cp36.cp37-none-win32.whl#md5=f486111e89542660ac29cfd791644b8b (from https://pypi.python.org/simple/pyqt5/) because it is not compatible with this Python
    Skipping https://pypi.python.org/packages/28/9e/ae20f544f99f6e5b28d806bf2c37b7069c8bb3fd7bdf84e6ed02ee853591/PyQt5-5.9-5.9.1-cp35.cp36.cp37-abi3-macosx_10_6_intel.whl#md5=bf07195ac2b81821ade2bc51762784c9 (from https://pypi.python.org/simple/pyqt5/) because it is not compatible with this Python
    Skipping https://pypi.python.org/packages/34/a7/ed3a2e7f74a7fcc112c1669968ca1be552f34d123bb4515a97520d69e8f3/PyQt5-5.6-cp35-cp35m-manylinux1_x86_64.whl#md5=787c331fb9b462a16f8526c964023c58 (from https://pypi.python.org/simple/pyqt5/) because it is not compatible with this Python
    Skipping https://pypi.python.org/packages/3a/c6/26270f5550f00920045c2f0b222a7d03d7a64382825c68bf0bb1a51d854c/PyQt5-5.9.2-5.9.3-cp35.cp36.cp37-abi3-manylinux1_x86_64.whl#md5=7af06d4bd8360b710e62e0c21e7f110d (from https://pypi.python.org/simple/pyqt5/) because it is not compatible with this Python
    Skipping https://pypi.python.org/packages/3b/08/32236a8fb73e4dfe51c8091a58ac24cc4987d1b12630a68939ebb6126d45/PyQt5-5.8-5.8.0-cp34.cp35.cp36.cp37-abi3-macosx_10_6_intel.whl#md5=73745a08f3cc1690922b6e21be341f43 (from https://pypi.python.org/simple/pyqt5/) because it is not compatible with this Python
    Skipping https://pypi.python.org/packages/3c/34/a768dacf653a883f57928ed971575c29f80b87db09a52c9b66724d6317a0/PyQt5-5.7-cp35-none-win_amd64.whl#md5=43f6617cd9d109ad334d55a1e2f8eecb (from https://pypi.python.org/simple/pyqt5/) because it is not compatible with this Python
    Skipping https://pypi.python.org/packages/40/f9/0be162881f4bdb6e828ee6db871310140a5278b4538e65716eae8b2d3dbd/PyQt5-5.6-cp35-cp35m-macosx_10_6_intel.whl#md5=51c8e78f58777eb71e0304ed8a35153d (from https://pypi.python.org/simple/pyqt5/) because it is not compatible with this Python
    Skipping https://pypi.python.org/packages/45/63/353403622330d335631ec3a3016f9e0a6a9519082488b1421003682f8ef4/PyQt5-5.9-5.9.1-cp35.cp36.cp37-abi3-manylinux1_x86_64.whl#md5=785f1aa6027e2b691b9117629657f104 (from https://pypi.python.org/simple/pyqt5/) because it is not compatible with this Python
    Skipping https://pypi.python.org/packages/4f/52/428af82f860dae1e08f8cb7f03399a9cc0f3756f2bd1b3e31ee749e126e4/PyQt5-5.8.2-5.8.0-cp35.cp36.cp37-abi3-manylinux1_x86_64.whl#md5=4c2c1577d8766b4ce6f328635b9556fe (from https://pypi.python.org/simple/pyqt5/) because it is not compatible with this Python
    Skipping https://pypi.python.org/packages/53/47/c3af02173da1c6887de05ec6201f8582eee4efcd9476d47739b42cb1a8b9/PyQt5-5.9-5.9.1-cp35.cp36.cp37-none-win_amd64.whl#md5=93e37890bcec8009ede4a39faaad96c6 (from https://pypi.python.org/simple/pyqt5/) because it is not compatible with this Python
    Skipping https://pypi.python.org/packages/55/08/a8cae8689ed928a6c7eba2e125b5cf96545995692d144d8b8f6471e47246/PyQt5-5.8.2-5.8.0-cp35.cp36.cp37-none-win_amd64.whl#md5=ab236f946ce9f7819a1e240b41e4d6ef (from https://pypi.python.org/simple/pyqt5/) because it is not compatible with this Python
    Skipping https://pypi.python.org/packages/5c/9b/c5a06dccbf7db02c83adca82321f4b8e97aaf088d73eb5fc52f4f1463399/PyQt5-5.7.1-5.7.1-cp34.cp35.cp36-abi3-macosx_10_6_intel.whl#md5=6f3d3496505a39c44aed09fca0850f2c (from https://pypi.python.org/simple/pyqt5/) because it is not compatible with this Python
    Skipping https://pypi.python.org/packages/5e/41/93bf8f289b476bc5d8cece7948a166cbcea761f5b1262e5fdd7076a5e02a/PyQt5-5.9.2-5.9.3-cp35.cp36.cp37-abi3-macosx_10_6_intel.whl#md5=926b3841d74d178171a2bbc0946fe2a4 (from https://pypi.python.org/simple/pyqt5/) because it is not compatible with this Python
    Skipping https://pypi.python.org/packages/67/ad/5ce6033bded3c39043a2c67cc46b3976427187f68c9dbaf5533884899086/PyQt5-5.9.2-5.9.3-cp35.cp36.cp37-none-win_amd64.whl#md5=87f07e234e60dfc3a81d8b59862fb258 (from https://pypi.python.org/simple/pyqt5/) because it is not compatible with this Python
    Skipping https://pypi.python.org/packages/71/c5/7e2a35682358f1be16f5f310e7ea3b64c5ecc168f83c3dc4d50dcee9f087/PyQt5-5.7-cp35-none-win32.whl#md5=f52427b4706eb4f9a371ef5d8c02c9ac (from https://pypi.python.or [... cut ...]
    requirement_set.prepare_files(finder, force_root_egg_info=self.bundle, bundle=self.bundle)
  File "/usr/lib/python3/dist-packages/pip/req.py", line 1178, in prepare_files
    url = finder.find_requirement(req_to_install, upgrade=self.upgrade)
  File "/usr/lib/python3/dist-packages/pip/index.py", line 277, in find_requirement
    raise DistributionNotFound('No distributions at all found for %s' % req)
pip.exceptions.DistributionNotFound: No distributions at all found for pyqt5 (from hue-plus)
```

kusti8, 2017-11-26:
What's your Python version and operating system? Sometimes it's easier to
install it from the system package manager if something like this comes up,
pyqt5 is weird, but I do it this way to prevent guessing on the user's OS

jakedowns, 2017-11-26:
OS: `Windows 10`
Shell Env: `Bash on Ubuntu on Windows`
Python: `2.7.6` | `3.4.3`
pip: `9.0.1`

jakedowns, 2017-11-26:
update: finally got it to install with python 3.6.0a2 & pip 9.0.1

hitting a different speed bump now tho:

[image]

even though i can't call it from command line, i *was* able to finally call it from my own custom python script (i'm brand new to python)

```
#!/usr/bin/env python3
from hue_plus import hue
import serial

ser = serial.Serial("COM3", 256000)
hue.fixed(ser, 0, 0, "FF0000")
```

gonna play around with some crazy custom animated ones next, will let you know how it goes :P

closed 2017-11-26 by jakedowns

jakedowns, 2017-11-28:
@kusti8 thanks again for a great tool, I was able to write a little thing to convert any gif into a series of "color rounds" for your custom animation api

https://twitter.com/jakedowns/status/935337310513659904

jakedowns, 2017-11-28:
some more custom modes i'm working on if you're interested:
https://cl.ly/0W3n2i263h3K