kusti8/hue-plus #29: "No Hue+ found" popup on start. Program works fine otherwise.

PurifyPioneer, 2018-01-11:
Hello,

I found an issue i didn't  find a soloution for yet.

The problem is when i start the program or when the program starts after booting, I get a popup telling me that no hue+ was found (hue+ is isntalled and working!). When i click "Ok" the program opens as expected, takes 1-2 seconds to load and applies the selected effect (i chose spectrum wave). The hue is connected to "COM3".

I am on Windows 10. Latest updates as of 11.01.18. If you need any further information i will provide it.

wemaxi, 2018-02-01:
i have exactly this problem as well. i can switch colors and everything works as much as i can tell but it's confusing!

PurifyPioneer, 2018-02-09:
I think i might have identified the issue.

Are you guys running your system (windows) in another language than english (mine's running in german)?

Regarding the issue/fix:

In the file /hue_plus/hue_ui.py in line 171 the script calls the get_port() method.
The output of that method is evaluated in the following lines and will cause the "No Hue+ found." message if the output of that method is 'None'.

I assume that the get_port() method (line 212) will always return 'None' on systems that are not using the english language.
The method checks the COM ports for "MCP2200", "USB Serial Device" and "USB Serial Port" and will return the corresponding COM port(usually COM3) were it found a device containing one of the three descriptions.
However on my machine (using german) the description is "Serielles USB-Gerät" which is missed and the method will then return 'None'
I was able to fix the issue for my machine by adding `or 'Serielles USB-Gerät' in port[1]` to the end of the if statement in line 215.

But i guess to resolve that issue fully we need to find a way to either get the output with the correct locale (english) or look for a more general description (maybe "COM3/COM4") ?

I'm not expirienced enough in python or tinkering with COM-devices to come up with a soloution for this unfortunately.

The program is still working because the script seems to default to the correct port (COM3 in my case) (line 169)

mwebi, 2018-02-09:
Indeed my Windows system language is german as well.

Funny enough I had a quick look at the code yesterday and also marked that as a potential source for the error but didn't debug it since I don't usually develop with python.

I totally agree that a better way to identifying devices than string comparison would be best, but would you mind pushing your fix and triggering a build in the meanwhile?

kusti8, 2018-02-09:
Yeah I'm going to remove the pop-up since it's annoying. I currently don't
have a good idea on how to recognize a device without revamping the code,
and I don't have time for that. The problem with hard coding COM3 is the
fact that can change based on different computers. It normally is COM3 but
I've had it be different depending on what is plugged in. I think CAM does
it by sending an initial message to all serial ports and seeing if they
answer, but I don't really like that solution. So I'll remove it tomorrow
and then add a new release.

mwebi, 2018-02-10:
Nice, looking forward to the new build

PurifyPioneer, 2018-02-13:
@kusti8 
In another issue i read that you goal was to "let us set the colors of the hue+ and forget about it".
the first part is excellent but i can't really forget about it because i need to start your software once on every boot to get the hue+ to return to the set colors. I do not need to interact with the software other thank opening and closing it rightaway.
Is this intended ?
Because it came to my might that it might be related to this issue/hue+ not beeing found.

kusti8, 2018-02-13:
This is intended. When the computer turns off, them the Hue+ turns off as
well, meaning that it loses your settings. What I mean is that my goal is
to not have a constant running program such as CAM, but something where you
can just set the colors and safely close it. The problem with the error
dialog is that I didn't know the serial port name would be different based
on locale.

PurifyPioneer, 2018-02-13:
Ok fine, thanks for the quick answer and the amazing work you are doing!

mwebi, 2018-03-04:
Any ETA on when you'll have the time to push that commit and do the build? Would be greatly appreciated.

kusti8, 2018-03-04:
I totally forgot about that. Sorry. I'll push a fix either tomorrow or the day after.

mwebi, 2018-03-17:
I've created a pull request now with the solution as suggested by PurifyPioneer. Would be great if you accepted that. Of course you can always come up with a different solution if you had something else in mind. But this should give us a build without the popup at least in the meanwhile.

closed 2018-03-17 by kusti8

kusti8, 2018-03-17:
Sorry for the delay. I've pushed new versions

mwebi, 2018-03-17:
No worries, thanks

commit 2473270f (2018-03-17): Remove no device found. Fixes #29