kusti8/hue-plus #33: "Serial port is invalid" for every option available.

Kaicooper, 2018-03-26:
Tried using any of the serial ports given with the software, and each time it tells me the serial port is invalid.
This is both for Linux (/dev/ttyACM0) and using COM3 or COM4 for Windows.

invalidwarranty, 2018-07-21:
Hey man I know this is really late, but for me in linux after unplugging and rep lugging in th internal usb headers and running "sudo hue_ui" instead of "hue_ui" it worked, with ttyACM0

mwweissmann, 2018-07-21:
On Linux you must have write-access to the serial port; on Debian and Ubuntu the serial ports belong to group _dialout_, so adding yourself to that group lets you stay away from having to use sudo all the time. I'm pretty sure similar groups exist on pretty all Linux distributions, just check _/dev/ttyACM0_.