kusti8/hue-plus #36: hue-plus wont start

apathyrecharge, 2018-04-20:
hey there, i'm getting a traceback error when trying to run your script

Traceback (most recent call last):
  File "/usr/bin/hue", line 11, in <module>
    sys.exit(main())
  File "/usr/lib/python3.6/site-packages/hue_plus/hue.py", line 156, in main
    raise InvalidCommand("No such mode")
hue_plus.hue.InvalidCommand: No such mode

uname -a
`Linux amala 4.16.3-1-ARCH #1 SMP PREEMPT Thu Apr 19 09:17:56 UTC 2018 x86_64 GNU/Linux`

i'm not sure what other information to provide, please let me know.

kusti8, 2018-04-20:
What is the command that you're using to run it?

apathyrecharge, 2018-04-21:
i have "hue &" in my openbox autostart which used to work fine

it doesnt run on autostart so i get the aforementioned output from running "sudo hue" in the terminal (running without sudo says it doesnt have permission to access the ports)

i didn't mention it in the OP because they seem unrelated, but since hue-plus stopped working after i upgraded packages "libnghttp2" and "tslib" it might be worth mentioning after all.

Slimmons, 2018-05-24:
Same issue on Ubuntu 18.04.

running command ```sudo hue```

```
Traceback (most recent call last):
  File "/usr/local/bin/hue", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.6/dist-packages/hue_plus/hue.py", line 156, in main
    raise InvalidCommand("No such mode")
hue_plus.hue.InvalidCommand: No such mode
```

Slimmons, 2018-05-24:
I looked into the code, and realized the mistake I made.  hue, is the command line mode.  Which is pretty cool, but I wasn't aware (I'm sure there's something documenting it, but I didn't read it).  So, when I typed in ```sudo hue breathing``` it gives me more information on how to use the tool properly.  Using the ```hue_ui``` command gave me what I was expecting.  This isn't a bug for me, it's a failure on my part to read documentation :).  Thanks for the awesome tool.