kusti8/hue-plus #51: no device in /dev/ttyACM0, where is it then?

Amplespencer, 2019-05-27:
lsusb claims NZXT is
Bus 001 Device 003: ID 1e71:1714 NZXT 
so where would that be mounted?

I really have never done anything like this in linux before

fiatcrush, 2019-06-01:
I have the same issue, any advice please? 

command "usb-devices" - Gives me the below data and I am not sure how to work out the port from that:

```
T:  Bus=01 Lev=01 Prnt=01 Port=04 Cnt=04 Dev#=  4 Spd=12  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1e71 ProdID=1714 Rev=02.00
S:  Manufacturer=NZXT.-Inc.
S:  Product=NZXT USB Device
S:  SerialNumber=87685F895156
C:  #Ifs= 1 Cfg#= 1 Atr=c0 MxPwr=100mA
I:  If#= 0 Alt= 0 #EPs= 2 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbhid
```

Nyeksenn, 2019-08-01:
have you tried executing the application as root?

flaire-project, 2019-08-11:
I've used this script on Ubuntu to find which port is mapped to which USB device:

```
#!/bin/bash

for sysdevpath in $(find /sys/bus/usb/devices/usb*/ -name dev); do
    (
        syspath="${sysdevpath%/dev}"
        devname="$(udevadm info -q name -p $syspath)"
        [[ "$devname" == "bus/"* ]] && continue
        eval "$(udevadm info -q property --export -p $syspath)"
        [[ -z "$ID_SERIAL" ]] && continue
        echo "/dev/$devname - $ID_SERIAL"
    )
done
```

A-Human-Person, 2019-08-21:
continue is broken in that script and it claims the answer is

/dev/bus/usb/001/005 which is wrong

A-Human-Person, 2019-08-21:
<class 'serial.serialutil.SerialException'>: 
Could not configure port: (25, 'Inappropriate ioctl for device')

rjpiston, 2019-08-25:
I'm running into the same issue. Not sure where to go next.

T:  Bus=03 Lev=01 Prnt=01 Port=07 Cnt=02 Dev#=  2 Spd=12  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1e71 ProdID=1714 Rev=02.00
S:  Manufacturer=NZXT.-Inc.
S:  Product=NZXT USB Device
S:  SerialNumber=497D34943131
C:  #Ifs= 1 Cfg#= 1 Atr=c0 MxPwr=100mA
I:  If#= 0 Alt= 0 #EPs= 2 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbhid

saltchicken, 2020-03-15:
What is your output for `dmesg | grep tty`?

abhi-iyer, 2020-08-26:
Has there been a resolution on this? I'm having the same problem - the NZXT hub connected to the comp's USB is not under `/dev/ttyACM0`