kusti8/hue-plus #53: NZXT HUE 2 Support

flaire-project, 2019-08-11:
Are there any plans for supporting the new HUE 2 controller? I've tried to access the controller on `/dev/bus/usb/001/003` but it crashed with a `Inappropriate ioctl for device` error.

acerspyro, 2019-12-23:
Would have been good to know that it doesn't support HUE 2, I've been fiddling it for hours.

CalcProgrammer1, 2019-12-29:
I just picked up a Hue 2 and documented the protocol for my own project.  Feel free to use my documentation to add support to this project.  The protocol isn't exactly the same and I haven't figured out everything yet, but it's simple enough and the effects are close to the Hue Plus.  It uses USB interrupt transfers rather than a serial device.

https://gitlab.com/CalcProgrammer1/OpenRGB/-/wikis/NZXT-Hue-2

Eiion, 2021-06-01:
> I just picked up a Hue 2 and documented the protocol for my own project. Feel free to use my documentation to add support to this project. The protocol isn't exactly the same and I haven't figured [...]

Do I see right that your version doens't support the "Ambilight" feature, yet?

CalcProgrammer1, 2021-06-01:
Ambilight isn't a feature of the hardware, it's a feature of the software.  OpenRGB doesn't have an effects engine built-in for doing reactive/synchronized effects like Ambilight or Audio Visualizer.

However, OpenRGB has a network-based SDK as well as a plugin system.  I just released OpenRGB version 0.6 which supports the OpenRGB Effects Plugin.  This plugin does have an Ambilight mode and will work with any OpenRGB device that has direct control mode (which NZXT Hue 2 does).

Eiion, 2021-06-01:
Very nice! I'll keep an eye on your project and will look deeper into it once I got the time.