kusti8/hue-plus #56: could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'

dulex123, 2019-10-05:
Any ideas why this happens?

David-Frick, 2019-10-06:
Same issue here. I would really like to use this software so hopefully this gets worked out!

Gooberpatrol66, 2020-01-05:
try modprobe cdc-acm

JakubBrom, 2020-01-19:
> try modprobe cdc-acm

Does'nt work for me

viniciusvivan, 2020-02-09:
I'm having the same problem here.

bcote94, 2020-02-16:
Having exact same problem as OP.

uchaudh, 2020-03-05:
Same problem me too! Did anyone get a solution?

saltchicken, 2020-03-15:
Does it work if you run `sudo hue fixed AAAAAA`?

What OS are you running?

What is your output for `dmesg |grep tty`

bcote94, 2020-03-16:
Hi @saltchicken ,

1. The command spat back the same error, `serial.serialutil.SerialException: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'`

2. Pop_OS 19.10

3. 
`[    0.000000] printk: console [tty0] enabled`
`[    0.370926] 00:04: ttyS0 at I/O 0x3f8 (irq = 4, base_baud = 115200) is a 16550A`

BoomWortel, 2020-04-11:
@bcote94 @saltchicken i have the exact same issue first i used OpenRGB but i it did not reconise my LED. 
the LED that im using ar the NZXT H500i case led in windows it works fine but in linux i cant get them to work in OpenRGB i can change my x52 kraken watercoller leds and my ram modules but not the led

1. The command spat back te same error, serial.serialutil.SerialException: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'

2. i use also Pop!_OS 19.10

3.
[    0.104640] printk: console [tty0] enabled
[    0.590103] 00:05: ttyS0 at I/O 0x3f8 (irq = 4, base_baud = 115200) is a 16550A

Frick-David, 2020-04-12:
@BoomWortel so for OpenRGB, some motherboards need to have a patch applied to the kernal first. Once I applied the patch, OpenRGB works like a charm.

Rpascar1, 2020-06-21:
> @BoomWortel so for OpenRGB, some motherboards need to have a patch applied to the kernal first. Once I applied the patch, OpenRGB works like a charm.

What is this patch - can you please provide more information about solving this?

abhi-iyer, 2020-08-26:
@Frick-David do you mean you got `hue_ui` working on linux?

dpawlows, 2020-11-13:
@BoomWortel Did you ever solve this? I'm running into the same issue. I've tracked down my smart device to be connected to /dev/bus/usb/003 but that port doesn't work either, so I'm missing something.

Sacerdos1990, 2020-12-08:
https://gitlab.com/CalcProgrammer1/OpenRGB
this worked for me.  at the end it says:
Run the application with ./OpenRGB
use sudo ./openrgb     
that is what worked for me.

Frick-David, 2020-12-08:
This issue should probably be closed. Like @Sacerdos1990 said, look at openRGB, liquidctl, and there are some others as well.