kusti8/hue-plus #59: Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-65xk2tzc/pyqt5/

ProPatte, 2020-04-19:
OS: Linux Mint 19.3 Cinnamon.

CLI Output:

root@mentos:/home/patte# sudo pip3 install hue_plus
Collecting hue_plus
  Using cached https://files.pythonhosted.org/packages/87/85/0b8f510d2a41478fa08635de340c046b2bd6835072706809bc47da4e651a/hue_plus-1.4.5-py35-none-any.whl
Requirement already satisfied: pyserial in /usr/lib/python3/dist-packages (from hue_plus)
Collecting pyqt5 (from hue_plus)
  Using cached https://files.pythonhosted.org/packages/4d/81/b9a66a28fb9a7bbeb60e266f06ebc4703e7e42b99e3609bf1b58ddd232b9/PyQt5-5.14.2.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/usr/lib/python3.6/tokenize.py", line 452, in open
        buffer = _builtin_open(filename, 'rb')
    FileNotFoundError: [Errno 2] No such file or directory: '/tmp/pip-build-672_6msj/pyqt5/setup.py'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-672_6msj/pyqt5/

DatLycan, 2020-04-19:
I resolved this issue by installing the python language package.

`pip3 install python-language-server`

ProPatte, 2020-04-21:
> I resolved this issue by installing the python language package.
> 
> `pip3 install python-language-server`

Unfortunately that didn't change anything. I get the same error.

RStadlmair, 2020-05-06:
I have the same issue, even with installing python-language-server, the error exists. Happy to provide logs for troubleshooting. What logs shall i deliver ?